sidddy/flora #2: Any plans to add retrieving the battery level?

asexauer, 2018-04-11:
I tried this, but I am new to BLE. You included in your code the characteristic BLEUUID uuid_version_battery("00001a02-0000-1000-8000-00805f9b34fb"); and I checked for this one but I am unsure what to do with the HEX Result.
Any help would be great.

sidddy, 2018-04-11:
HI,

yes, I'll add the battery level at some point... probably within the next weeks. I'll update this issue once it's done.

bern6666, 2018-04-22:
I like your implementation of the Xiaomi Mi Plant Sensor very much, because it is written in such a clear style.
I would like to create a similar implementation for the Xiaomi BLE Thermometer MJ_HT_V1. It is a bit different as it uses notification. I can solve that, but the second time in the loop part it loses connection to the characteristic and crashes. Can you give me an advice for this?

As for your implementation of flora.ino:
1) Your sketch uses 107% of the capacity on my computer with nodemcu-32S and does not compile. I have to leave out WIFI and MQTT. Then it uses 83% and runs smoothly. How do you manage that it uses only 90% of the capacity with wifi and mqtt on your device?   
2) On line 102 "float temp = (val[0] + val[1] * 256) / 10.0;" you should divide by the float 10.0, otherwise you will get an integer division (26.00 instead of 26.2).
3) What is the purpose of lines 72 to 75? Can you explain please? Where did you find information about this?  
  pRemoteCharacteristic = pRemoteService->getCharacteristic(uuid_write_mode);
  uint8_t buf[2] = {0xA0, 0x1F};
  pRemoteCharacteristic->writeValue(buf, 2, true);

Kind regards
Bernhard

sidddy, 2018-04-23:
@bern6666 
regarding your questions;
1) A few months ago, I managed to fit the flora code into the available 1.3MB in the default ESP32 arduino partition layout. Unfortunately, updates to the ESP32 core library made the sketch too big meanwhile. A workaround would be to switch to another partition layout and increase the max build size.  Recent updates to the esp32 core enable manually switching to another partition layout (e.g. min_spiffs for minimized spiffs partition size). You'll still have to change your max build size in boards.txt. See commit https://github.com/espressif/arduino-esp32/commit/3235a2f38c2bd2edc080967053c7bef63aab9321 in my fork of the esp32 core library.

2) Good point, I'll change that.

3) For recent xiaomi mi flora firmwares, we have to enable the  data retrieval by sending two magic bytes to the specified UUID. I got this info from https://github.com/open-homeautomation/miflora/blob/master/miflora/miflora_poller.py#L72 and don't really have a clue what I send here... ;-)

bern6666, 2018-04-23:
Thanks a lot for your quick and kind answer. I did not fully understand your explanation of issue 1, but I found a recently published recommendation of changing the partition size which I think is what you proposed: 
https://desire.giesecke.tk/index.php/2018/04/20/change-partition-size-arduino-ide/

sidddy, 2018-04-23:
@bern6666 yes, that's basically what I meant :)

sidddy, 2018-04-23:
@asexauer: commit 5f028140358ea5a499b45f2ec38a138627088315 adds retrieval of the battery value and pushes it via mqtt. closing this issue.

closed 2018-04-23 by sidddy

asexauer, 2018-04-24:
Great to news.. THX I give it a try.

asexauer, 2018-04-24:
Just a little feedback. Works like a charm. I used your code to pull multiple devices with an esp32 and to publish the data to thingspeak instead of an mqtt server.