sidddy/flora #5: WiFi connect does not work

oetti77, 2018-06-25:
Hi sidddy,

i've compiled the code successfully, but i do not get a working wifi-connection. I've tried several wifi settings without any success. 

To eliminate hardware issues i've tested this code ( https://techtutorialsx.com/2017/06/29/esp32-arduino-getting-started-with-wifi/ ) and could connect immediately. 

I'm not sure how to debug this, so any help is appreciated :-)

Thanks
Chris

oetti77, 2018-06-25:
serial output is "Connecting WiFi..." until "Going to sleep". 
I do not see any connection attemps in the wifi logs (mikrotik).

board: Lolin32 01.0.0 (esp-wroom-32 chipset)

sidddy, 2018-07-09:
I'm not able to reproduce this issue... as far as I can see, the c"working wlan code" you referred to is using the same wifi calls as I do... can you enable wlan debug information and check serial console again?

sidddy, 2018-07-09:
I was referring to debugging on the ESP32 side: In arduino, you could change your board to ESP32 Dev Module, which would give you an additional "Debug" menu entry. Set core debug level to the highest one and try again.

Your issue description sounds much like https://github.com/espressif/arduino-esp32/issues/1212 . Maybe you can try the solution proposal given in the latest comment.

oetti77, 2018-07-09:
Indeed, after "patching" your code with the workaround described [here](https://github.com/espressif/arduino-esp32/issues/1212) everything is working now.

Thank you, and keep up the good work!

closed 2018-07-09 by oetti77

palsbo, 2018-07-12:
I have a similar problem. Connecting to WiFi can last forever. Worst with DHCP but also fixed IP takes long time.
I am running a web-radio and once connected, some streams are slower than others,  causing drop-outs from time to time (most in the beginning). This has nothing to do with bit-rate; even low bit-rate streams have problems.
I suspect, that the TCP protocol has a bug or a weakness as the problem is the same for all kinds of HTTP/WiFi clients.
I tried the tricks mentioned above with no effect (also only affects connection - if any) 
I use Arduino IDE. Internet speed is more than 100mb/s

Any suggestions are appreciated?