sidddy/flora #8: Support for Xiaomi Mijia Bluetooth Temperature Humidity Sensor LCD Screen Digital Thermometer Hygrometer

tunip, 2018-07-12:
Hi,

great work for the flora client!

Would you be interessted to implement or write a separate version for the Xiaomi Mijia Bluetooth Temperature Humidity Sensor LCD Screen Digital Thermometer Hygrometer?
(https://www.banggood.com/de/Xiaomi-Mijia-Bluetooth-Thermometer-Hygrometer-with-LCD-Screen-Magnetic-Suction-Wall-Stickers-p-1232396.html)

I have a few of them. Here are some relevant information.

`$ gatttool -b $mac --primary`
```
attr handle = 0x0001, end grp handle = 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle = 0x0008, end grp handle = 0x000b uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle = 0x000c, end grp handle = 0x0015 uuid: 226c0000-6476-4566-7562-66734470666d (service for sensor_data)
attr handle = 0x0016, end grp handle = 0x0019 uuid: 0000180f-0000-1000-8000-00805f9b34fb (service for battery_data)
attr handle = 0x001a, end grp handle = 0x0024 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle = 0x0025, end grp handle = 0x002c uuid: 00001530-1212-efde-1523-785feabcd123
attr handle = 0x002d, end grp handle = 0xffff uuid: 0000fe95-0000-1000-8000-00805f9b34fb
```
`$ gatttool -b $mac --characteristics`
```
handle: 0x0002, char properties: 0x0a, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0009, char properties: 0x20, char value handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x10, char value handle: 0x000e, uuid: 226caa55-6476-4566-7562-66734470666d  (sensor_data)
handle: 0x0012, char properties: 0x18, char value handle: 0x0013, uuid: 226cbb55-6476-4566-7562-66734470666d
handle: 0x0017, char properties: 0x12, char value handle: 0x0018, uuid: 00002a19-0000-1000-8000-00805f9b34fb  (battery_data)
handle: 0x001b, char properties: 0x02, char value handle: 0x001c, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x001d, char properties: 0x02, char value handle: 0x001e, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x001f, char properties: 0x02, char value handle: 0x0020, uuid: 00002a25-0000-1000-8000-00805f9b34fb
handle: 0x0021, char properties: 0x02, char value handle: 0x0022, uuid: 00002a27-0000-1000-8000-00805f9b34fb
handle: 0x0023, char properties: 0x02, char value handle: 0x0024, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x0026, char properties: 0x04, char value handle: 0x0027, uuid: 00001532-1212-efde-1523-785feabcd123
handle: 0x0028, char properties: 0x18, char value handle: 0x0029, uuid: 00001531-1212-efde-1523-785feabcd123
handle: 0x002b, char properties: 0x02, char value handle: 0x002c, uuid: 00001534-1212-efde-1523-785feabcd123
handle: 0x002e, char properties: 0x18, char value handle: 0x002f, uuid: 00000001-0000-1000-8000-00805f9b34fb
handle: 0x0031, char properties: 0x02, char value handle: 0x0032, uuid: 00000002-0000-1000-8000-00805f9b34fb
handle: 0x0033, char properties: 0x02, char value handle: 0x0034, uuid: 00000004-0000-1000-8000-00805f9b34fb
handle: 0x0035, char properties: 0x08, char value handle: 0x0036, uuid: 00000010-0000-1000-8000-00805f9b34fb
handle: 0x0037, char properties: 0x0a, char value handle: 0x0038, uuid: 00000013-0000-1000-8000-00805f9b34fb
handle: 0x0039, char properties: 0x0a, char value handle: 0x003a, uuid: 00000014-0000-1000-8000-00805f9b34fb
```
`$ gatttool -b $mac --char-desc`
```
...
handle: 0x0010, uuid: 00002902-0000-1000-8000-00805f9b34fb (sensor_write value 0010)
...
```

Get information with gatttool:
**Temperature & Humidity**
`$ gatttool -b $mac --char-write-req --handle 0x10 --value=0100 --listen`
```
Characteristic value was written successfully
Notification handle = 0x000e value: 54 3d 32 33 2e 39 20 48 3d 35 32 2e 34 00
Notification handle = 0x000e value: 54 3d 32 34 2e 30 20 48 3d 35 32 2e 33 00
Notification handle = 0x000e value: 54 3d 32 33 2e 39 20 48 3d 35 32 2e 33 00
```
`$ echo "54 3d 32 33 2e 39 20 48 3d 35 32 2e 34 00" | xxd -r -p`
```T=23.9 H=52.4```

**Battery:**
`$ gatttool -b $mac --char-read --handle 0x18`
```
Characteristic value/descriptor: 64
```
`$ echo "ibase=16; 64" | bc`
```
100
```

sidddy, 2018-07-15:
HI,

thanks for your input, but unfortunately I don't plan to add support for Mijia sensors. Also check issue #6 .

closed 2018-07-15 by sidddy

aHVzY2g, 2018-07-15:
Hey tunip, I found some working code for the sensor.

So far it only grabs the temprature, and it crashes realy often. I will get back to you and send you a link to the code.

aHVzY2g, 2018-09-19:
OpenMQTTGateway is the solution @tunip

tunip, 2018-09-20:
@aHVzY2g Thx for the hint.