sidddy/flora #9: Crashes for first try, then stuck at connecting

carnifex82, 2018-08-17:
Hi there,

after changing the partition table, I got the sketch successfully uploaded to my ESP32 dev board. 

However, it fails to read any miflora data. Often it crashes a few time before performing a connection, but even then nothing else happens before it goes to sleeo mode:

`Starting Flora client...
Connecting WiFi...
.....
WiFi connected
Attempting MQTT connection...connected
Forming a connection to Flora device at c4:7c:8d:66:51:40
 - Connected to Flora
abort() was called at PC 0x401a780f on core 1

Backtrace: 0x40095324:0x3ffdaaa0 0x40095527:0x3ffdaac0 0x401a780f:0x3ffdaae0 0x401a7856:0x3ffdab00 0x4019513b:0x3ffdab20 0x400d31b1:0x3ffdab40 0x400d1627:0x3ffdaba0 0x400d1d91:0x3ffdac70 0x401af3c6:0x3ffdacc0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Starting Flora client...
Connecting WiFi...
.....
WiFi connected
Attempting MQTT connection...connected
Forming a connection to Flora device at c4:7c:8d:66:51:40
Goi`

I tried two miflora devices (that work in the official app) and also changed the sleep wait to 120.

Any ideas?

sidddy, 2018-08-26:
Hi,

could you provide a full stack trace of the error via the EspExceptionDecoder? Check https://github.com/me-no-dev/EspExceptionDecoder .

Thanks,
Sidddy

sidddy, 2018-08-26:
potentially related to #10

carnifex82, 2018-08-26:
Now, that's interesting. I installed the exception decoder tool and connected the esp32 via USB. I checked the console and copied the adress. When I opened the tool, it asked me for the .elf file. A quick google told me that I need to compile the sketch again. Did so, uploaded ... and it's not crashing anymore?

But sadly, it's not connecting, either. It gets into sleep before retrieving any data.

SLEEP_WAIT is set to 120  

I guess this is a new issue then?

Starting Flora client...
Connecting WiFi...
.....
WiFi connected
Attempting MQTT connection...connected
Forming a connection to Flora device at c4:7c:8d:66:51:40
Goi

carnifex82, 2018-08-26:
Oh wait .. I changed the MAC Adress to the other miFlora just to be sure. Now it's crashing again.

Starting Flora client...
Connecting WiFi...
......
WiFi connected
Attempting MQTT connection...connected
Forming a connection to Flora device at c4:7c:8d:65:5a:d5
 - Connected to Flora
abort() was called at PC 0x401a780f on core 1

Backtrace: 0x40095324:0x3ffdaaa0 0x40095527:0x3ffdaac0 0x401a780f:0x3ffdaae0 0x401a7856:0x3ffdab00 0x4019513b:0x3ffdab20 0x400d31b1:0x3ffdab40 0x400d1627:0x3ffdaba0 0x400d1d91:0x3ffdac70 0x401af3c6:0x3ffdacc0

Rebooting...

With the decoder it says:

Decoding stack results
```
0x40095324: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x40095527: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x401a780f: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401a7856: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4019513b: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x400d31b1: BLEClient::getService(BLEUUID) at C:\Users\xfleog\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\BLE\src\BLEClient.cpp line 340
0x400d1627: getSensorData(BLEAddress, bool) at C:\Users\xfleog\Documents\Arduino\flora/flora.ino line 66
0x400d1d91: setup() at C:\Users\xfleog\Documents\Arduino\flora/flora.ino line 214
0x401af3c6: loopTask(void*) at C:\Users\xfleog\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\main.cpp line 15

```

lociii, 2018-10-28:
Seems to be related to https://github.com/nkolban/esp32-snippets/issues/419#issuecomment-373467202

sidddy, 2018-10-31:
Agreed. The stacktrace shows that we can's really do anything about this crash in flora, it's an issue in the BLE coding. Please open an issue over there if the problem still exists.

closed 2018-10-31 by sidddy

lociii, 2018-10-31:
That's totally right. I would like to keep this issue open or add a remark to the readme for future users.
Unfortunately this project is currently not in a working state even if it's totally not your (the project owners) fault.
This fact shouldn't be hidden to your users.