sidddy/flora #10: keeps crashing

Warfacenl, 2018-08-22:
- Connected to Flora
abort() was called at PC 0x401a7813 on core 1

Backtrace: 0x40095324:0x3ffdab30 0x40095527:0x3ffdab50 0x401a7813:0x3ffdab70 0x401a785a:0x3ffdab90 0x4019513f:0x3ffdabb0 0x400d31b1:0x3ffdabd0 0x400d1627:0x3ffdac30 0x400d1d91:0x3ffdad00 0x401af3ca:0x3ffdad50

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Starting Flora client...
Connecting WiFi...
....
WiFi connected
Attempting MQTT connection...connected
Forming a connection to Flora device at xx:xx:xx:xx:xx:xx
 - Connected to Flora
abort() was called at PC 0x401a7813 on core 1

Backtrace: 0x40095324:0x3ffdaaa0 0x40095527:0x3ffdaac0 0x401a7813:0x3ffdaae0 0x401a785a:0x3ffdab00 0x4019513f:0x3ffdab20 0x400d31b1:0x3ffdab40 0x400d1627:0x3ffdaba0 0x400d1d91:0x3ffdac70 0x401af3ca:0x3ffdacc0

Rebooting...
ets Jun  8 2016 00:22:57

sidddy, 2018-08-26:
Hi,

could you provide a full stack trace of the error via the EspExceptionDecoder? Check https://github.com/me-no-dev/EspExceptionDecoder .

Thanks,
Sidddy

sidddy, 2018-10-31:
closing due to lack of feedback

closed 2018-10-31 by sidddy