sidddy/flora #13: Failed to find data service

wackerr, 2018-11-11:
Hi,

I have been testing both the sidddy / Flora and Lociii / Flora but I runt into issues.
With the Lociii fork I get this after it connects to the flora:
- Failed to find data service

Anyone that got a clue on what to do?

Thanks

sandrisj, 2018-11-11:
I have the same issue. Any ideas how to fix it?

sidddy, 2018-11-16:
Please check the following:
- did you enter the correct MAC address for the device?
- is the device in close distance to the esp32? At least for testing purposes, you should have the sensor max 1m away from the esp32. Once the connection is known to work, you can test how far away you can position the sensor.

If this does not help, please provide the serial log of your failed connection attempt. Note that the changes from Locii were merged now, so please pull the master branch for your tests.

wackerr, 2018-11-16:
Hi,
I have now used the master branch but same issue.
5:52:55.860 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:52:55.860 -> configsip: 0, SPIWP:0xee
15:52:55.860 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:52:55.860 -> mode:DIO, clock div:1
15:52:55.860 -> load:0x3fff0018,len:4
15:52:55.860 -> load:0x3fff001c,len:952
15:52:55.860 -> load:0x40078000,len:6084
15:52:55.860 -> load:0x40080000,len:7936
15:52:55.860 -> entry 0x40080310
Initialize BLE client...
Connecting to WiFi...
........
15:53:01.960 -> WiFi connected
15:53:01.960 -> 
15:53:01.960 -> Connecting to MQTT...
MQTT connected
15:53:01.994 -> 
15:53:01.994 -> Processing Flora device at c4:7c:8d:67:65:19 (try 1)
- Connection successful
- Failed to find data service

sidddy, 2018-11-18:
That's weird, it looks like the connection to the device itself is successful, but the device does not offer the required data service for the connection. Could you check with some bluetooth analysis device if your flora sensor actually provides the data service with UUID 00001204-0000-1000-8000-00805f9b34fb? Easiest way is probably to use an app on your mobile phone to check this, e.g. nRF Connect.

wackerr, 2018-11-19:
Seems like it is there. 

[image]

sidddy, 2018-11-25:
Hi,

could you try the following?
1) Please flash flora with "Core Debug Level" set to "Verbose":
[image]
2) Check the content from Serial Monitor after reset. You should see lots of log entries from ESP32 core now. We're looking for entries with content "GATT Event: ...". If you see these, debug log was successfully enabled :-)
3) Please attach the complete content of the serial monitor here.

BR,
Sidddy

wackerr, 2018-11-25:
Here it comes:
ets Jun  8 2016 00:22:57
18:59:44.326 -> 
18:59:44.326 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:59:44.326 -> configsip: 0, SPIWP:0xee
18:59:44.326 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:59:44.326 -> mode:DIO, clock div:1
18:59:44.326 -> load:0x3fff0018,len:4
18:59:44.326 -> load:0x3fff001c,len:952
18:59:44.326 -> load:0x40078000,len:6084
18:59:44.326 -> load:0x40080000,len:7936
18:59:44.326 -> entry 0x40080310
Initialize BLE client...
[D][BLEDevice.cpp:444] setPower(): >> setPower: 7
18:59:46.426 -> [D][BLEDevice.cpp:449] setPower(): << setPower
18:59:46.426 -> Connecting to WiFi...
18:59:46.426 -> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
18:59:46.426 -> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
.[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
......[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
18:59:50.494 -> [D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: -------------------, MASK: ------------------, GW: --------------------
18:59:50.494 -> .
18:59:50.494 -> WiFi connected
18:59:50.494 -> 
18:59:50.494 -> Connecting to MQTT...
18:59:50.494 -> MQTT connected
18:59:50.494 -> 
18:59:50.494 -> Processing Flora device at c4:7c:8d:67:65:19 (try 1)
18:59:50.494 -> [D][BLEDevice.cpp:54] createClient(): >> createClient
18:59:50.494 -> [D][BLEDevice.cpp:60] createClient(): << createClient
18:59:50.494 -> [D][BLEClient.cpp:91] connect(): >> connect(c4:7c:8d:67:65:19)
18:59:50.494 -> [D][BLEClient.cpp:74] clearServices(): >> clearServices
18:59:50.494 -> [D][BLEClient.cpp:81] clearServices(): << clearServices
18:59:50.494 -> [D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
18:59:50.494 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
18:59:50.494 -> [D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
18:59:52.748 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
18:59:52.748 -> [D][BLEUtils.cpp:1407] dumpGattClientEvent(): [conn_id: 0, remote_bda: c4:7c:8d:67:65:19]
18:59:52.748 -> [D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
18:59:52.748 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
18:59:52.748 -> [D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, remote_bda: c4:7c:8d:67:65:19, mtu: 23]
18:59:52.748 -> [D][BLEClient.cpp:123] connect(): << connect(), rc=1
18:59:52.748 -> - Connection successful
18:59:52.748 -> [D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001204-0000-1000-8000-00805f9b34fb
18:59:52.748 -> [D][BLEClient.cpp:358] getServices(): >> getServices
18:59:52.748 -> [D][BLEClient.cpp:74] clearServices(): >> clearServices
18:59:52.748 -> [D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
19:00:01.625 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
19:00:01.625 -> [D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_TIMEOUT, conn_id: 0, remote_bda: c4:7c:8d:67:65:19]
19:00:01.625 -> [D][BLEClient.cpp:374] getServices(): << getServices
19:00:01.625 -> [D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CLOSE_EVT
19:00:01.625 -> [D][BLEClient.cpp:339] getService(): << getService: not found
19:00:01.625 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CLOSE_EVT
19:00:01.625 -> [D][BLEUtils.cpp:1392] dumpGattClientEvent(): [status: ESP_GATT_OK, reason:ESP_GATT_CONN_TIMEOUT, conn_id: 0]
19:00:01.625 -> - Failed to find data service
19:00:01.625 -> [D][BLEClient.cpp:133] disconnect(): >> disconnect()
19:00:01.625 -> [D][BLEClient.cpp:141] disconnect(): << disconnect()
19:00:01.625 -> [D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_UNREG_EVT
19:00:01.625 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_UNREG_EVT
Processing Flora device at c4:7c:8d:67:65:19 (try 2)
19:00:02.642 -> [D][BLEDevice.cpp:54] createClient(): >> createClient
19:00:02.642 -> [D][BLEDevice.cpp:60] createClient(): << createClient
19:00:02.642 -> [D][BLEClient.cpp:91] connect(): >> connect(c4:7c:8d:67:65:19)
19:00:02.642 -> [D][BLEClient.cpp:74] clearServices(): >> clearServices
19:00:02.642 -> [D][BLEClient.cpp:81] clearServices(): << clearServices
19:00:02.642 -> [D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
19:00:02.642 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
19:00:02.642 -> [D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[... 40 lines cut ...]
19:00:45.643 -> [D][BLEClient.cpp:123] connect(): << connect(), rc=1
19:00:45.643 -> - Connection successful
19:00:45.643 -> [D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001204-0000-1000-8000-00805f9b34fb
19:00:45.643 -> [D][BLEClient.cpp:358] getServices(): >> getServices
19:00:45.643 -> [D][BLEClient.cpp:74] clearServices(): >> clearServices
[... cut ...]
19:00:52.780 -> [D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_UNREG_EVT
19:00:52.780 -> [D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_UNREG_EVT
WiFi disonnected
19:00:55.287 -> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 3 - STA_STOP
19:00:55.287 -> [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 3 - STA_STOP
19:00:55.287 -> [E][WiFiClient.cpp:338] write(): 113
19:00:55.287 -> MQTT disconnected
19:00:55.287 -> Going to sleep now.

sandrisj, 2018-11-25:
I have the same issue:

`Connecting to MQTT...
MQTT connected

Processing Flora device at c4:7c:8d:67:0f:a5 (try 1)
[D][BLEDevice.cpp:54] createClient(): >> createClient
[D][BLEDevice.cpp:60] createClient(): << createClient
[D][BLEClient.cpp:91] connect(): >> connect(c4:7c:8d:67:0f:a5)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1407] dumpGattClientEvent(): [conn_id: 0, remote_bda: c4:7c:8d:67:0f:a5]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, remote_bda: c4:7c:8d:67:0f:a5, mtu: 23]
[D][BLEClient.cpp:123] connect(): << connect(), rc=1
- Connection successful
[D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001204-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:358] getServices(): >> getServices
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_TIMEOUT, conn_id: 0, remote_bda: c4:7c:8d:67:0f:a5]
[D][BLEClient.cpp:374] getServices(): << getServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CLOSE_EVT
[D][BLEClient.cpp:339] getService(): << getService: not found
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CLOSE_EVT
[D][BLEUtils.cpp:1392] dumpGattClientEvent(): [status: ESP_GATT_OK, reason:ESP_GATT_CONN_TIMEOUT, conn_id: 0]
- Failed to find data service
[D][BLEClient.cpp:133] disconnect(): >> disconnect()
[D][BLEClient.cpp:141] disconnect(): << disconnect()
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_UNREG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_UNREG_EVT
Processing Flora device at c4:7c:8d:67:0f:a5 (try 2)
[D][BLEDevice.cpp:54] createClient(): >> createClient
[D][BLEDevice.cpp:60] createClient(): << createClient
[D][BLEClient.cpp:91] connect(): >> connect(c4:7c:8d:67:0f:a5)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1407] dumpGattClientEvent(): [conn_id: 0, remote_bda: c4:7c:8d:67:0f:a5]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, remote_bda: c4:7c:8d:67:0f:a5, mtu: 23]
[D][BLEClient.cpp:123] connect(): << connect(), rc=1
- Connection successful
[D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001204-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:358] getServices(): >> getServices
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_TIMEOUT, conn_id: 0, remote_bda: c4:7c:8d:67:0f:a5]
[D][BLEClient.cpp:374] getServices(): << getServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CLOSE_EVT
[D][BLEClient.cpp:339] getService(): << getService: not found
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CLOSE_EVT
- Failed to find data service
[D][BLEUtils.cpp:1392] dumpGattClientEvent(): [status: ESP_GATT_OK, reason:ESP_GATT_CONN_TIMEOUT, conn_id: 0]
[D][BLEClient.cpp:133] disconnect(): >> disconnect()
[... 5 lines cut ...]
[D][BLEDevice.cpp:60] createClient(): << createClient
[D][BLEClient.cpp:91] connect(): >> connect(c4:7c:8d:67:0f:a5)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[... cut ...]
[D][BLEDevice.cpp:60] createClient(): << createClient
[D][BLEClient.cpp:91] connect(): >> connect(c4:7c:8d:67:0f:a5)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_ERROR, app_id: 0x0]
`

wackerr, 2018-12-08:
Hi,
Is there anything else we can do to help? Would be really cool to get this working before christmas(was planned to be a gift). Thanks

slu125, 2018-12-09:
Got the same problem, but sometimes it works, after I reset for a couple of times. It also doesn't work reliable when the flower sensor is right next to the esp32. Any tips, would be very helpful :-)

sidddy, 2019-07-11:
@wackerr @sandrisj Does this issue still occur with arduino-esp32 1.0.2? I'm still unable to reproduce this problems and the logs don't really give a hint...

ng2dev, 2019-09-20:
Hi siddy - yes the issue does still occur with the v1 esp32 board - I have the same issue.

obn100, 2019-09-23:
Thanks for your efforts Sidddy,
same issue here with a HHCC Flower Care sensor, Firmware 3.2.1

e6on, 2019-09-28:
I have same issue with Flower Care sensor (FW 3.2.1) and Adafruit HUZZAH32 - ESP32 Feather.

e6on, 2019-09-29:
I think I discovered why the data service is not found. I added this line `Serial.println(serviceUUID.toString().c_str());` to the program. It prints the UUID and as you can see the UUID is missing 2 characters (fb) in the end, but I don't know how to fix it.
`Processing Flora device at c4:7c:8d:6a:e2:1b (try 1)
- Connection successful
- Failed to find data service
00001204-0000-1000-8000-00805f9b34
`

ng2dev, 2019-09-29:
If I pass it defined like so:
const char* serviceUUID = "00001204-0000-1000-8000-00805f9b34fb"
to getService the serial println shows the correct string.
However, the lib still fails to find the data service as it is cast to this BLEUUID object.
I suspect that the error occurs in this library here.
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEUUID.h

obn100, 2019-09-29:
Thank you @e6on and @ng2dev for your analysis. Now it works for me :-)
In Arduino IDE I searched for ESP32 BLE and added the library by Neil Kolban / Version 1.0.1 - which is also stated at the main page of his project.
Sketch -> Include Library -> Manage Libraries...

Instead of this:
_#include "BLEDevice.h"_
I now added that library:
_#include <BLEDevice.h>_

Now I can see the sensor outputs.

ng2dev, 2019-09-29:
Ha great work guys - open source at its finest

ChrisSutton123, 2020-03-04:
I have a Miflora plant sensor working with well with ESP32 using this s/w.
I just took delivery of 5 VegTrug sensors.
They connect to okay, but the program cannot find the data service.
I am in a position to do any testing necessary