sidddy/flora #16: MQTT gives socket error

pedrogense, 2018-12-04:
Im having this error in my mosquitto broker on home assistant

1543960126: New client connected from 192.168.1.61 as xiaomiflora-client (c1, k15, u'peke').
1543960148: Client xiaomiflora-client has exceeded timeout, disconnecting.
1543960148: Socket error on client xiaomiflora-client, disconnecting.

My config:
// sleep between to runs in seconds
#define SLEEP_DURATION 60 * 60
// emergency hibernate countdown in seconds
#define EMERGENCY_HIBERNATE 1 * 60
// how often should the battery be read - in run count
#define BATTERY_INTERVAL 2
// how often should a device be retried in a run when something fails
#define RETRY 5

and for mqtt:
const int     MQTT_RETRY_WAIT = 5000;

Can someone point me in the right direction

ricreis394, 2019-01-29:
I'm having the same problem, did you fix that?

ricreis394, 2019-01-30:
nevermind, restart the esp32 and worked again

sidddy, 2019-01-30:
@pedrogense: Do these messages actually have any functional impact or do you still see the measurements? It looks to me like the ESP32 is going to deep sleep and mosquitto eventually complains about a timeout. We could properly log off from mosquitto before going to sleep, but that would potentially waste precious battery power...

indy55, 2019-02-15:
I am also experiencing the same issue. Only changes to config are MACS for sensors, WIFI credentials and MQTT credentials.

1550212766: New connection from 3.14.1.191 on port 1883.
1550212766: New client connected from 3.14.1.191 as miflora-client (c1, k15, u'').
1550212766: Sending CONNACK to miflora-client (0, 0)
1550212787: Client miflora-client has exceeded timeout, disconnecting.
1550212787: Socket error on client miflora-client, disconnecting.

No data from the sensors is published to MQTT.

pedrogense, 2019-04-01:
Let me know what comes out of it.

PhipsSti, 2019-04-02:
Seems to only partly fix the issue. I will debug the problem a bit more in detail.

PhipsSti, 2019-04-03:
Did some debugging. Seems that the communication to the flora-devices gets stuck quite often. This causes that the forced hibernate triggers and nothing is being disconnected gracefully. 1 out of 15 successfull reads for both configured floras is not such a good rate.

pedrogense, 2019-04-03:
Thats not a good rate, i can see that it not always updates, i have mutiple esp's on different levels in my house.
Do you think you can fix it?

PhipsSti, 2019-04-04:
I am not sure. At the moment I am thinking about starting a new project (with a bit of a different idea) for this task. But before doing so I will dig a bit around and give it one last try.

PhipsSti, 2019-04-06:
Just started my own implementation and noticed that I am facing the same problem. Reason seems to be the concurrent connection to wifi. So if we want to have successful reads we have to change the concept a bit.

pedrogense, 2019-04-06:
Cannot it be connect tru wifi all the time?

PhipsSti, 2019-04-07:
In case of an active connection to WiFi the ESP will quite often get stuck during reading the Floras. This leads to the forced hibernate and results in the harsh disconnect from MQTT and WiFi.

pedrogense, 2019-04-09:
I already stopped to use this, i recently found esphome integration and it works how it looks for now.

closed 2019-07-11 by sidddy