sidddy/flora #17: Strange value

Sildenafil99, 2019-04-06:
Hi,
I have uploaded the sketch on an esp32 board (wroom32), but i have strange value from my 3 miflora.
Where Is the problem?
The configuration Is stock, I have only changed the wifi, mqtt and mac info.
http://imgur.com/gallery/0zGUniZ
http://imgur.com/gallery/iNlmb3g

sidddy, 2019-07-11:
Could you connect the esp32 to serial console and paste the serial output here?

lhanneus, 2020-04-17:
Same problem for me, when I have several flora

Hex: 0 DC FC 3F 14 DD FC 3F 48 69 FF 3F 19 0 0 0  
-- Temperature: -921.60
-- Moisture: 63
-- Light: 5183
-- Conductivity: 26952

I am pretty sure it should be around 23°C

No problem when the same flora is connected alone : 
Hex: EA 0 0 4B 19 0 0 0 0 0 2 3C 0 FB 34 9B  
-- Temperature: 23.40
-- Moisture: 0
-- Light: 6475
-- Conductivity: 0

lhanneus, 2020-04-17:
Other errors messages :
Hex: 0 DC FC 3F 14 DD FC 3F B0 6A FF 3F 19 0 0 0  
-- Temperature: -921.60
-- Moisture: 63
-- Light: 5183
-- Conductivity: 27312

or

Hex: 0 DC FC 3F 14 DD FC 3F EC 68 FF 3F 19 0 0 0  
-- Temperature: -921.60
-- Moisture: 63
-- Light: 5183
-- Conductivity: 26860

lhanneus, 2020-04-17:
I think they are Bluetooth collision when there are several bluetooth devices responding. I'm trying to increase the time between each device access . 

It seems clean with that solution. I'm trying for a longer period to see if it fix the problem. And I'll update here.

cateyes1963, 2020-04-17:
this has solved all problems:

if (temperature > 200 || temperature < -30 || conductivity > 2000) { 
    Serial.println("-- Unreasonable values received, skip publish +++++++++++++++++++++++++++++");
    return false;
  }