sidddy/flora #18: MQTT server received no message

roc634, 2019-04-14:
I use an esp32 to communicate with the host of my HomeAssistant. I can receive the message from another esp8266, but not from this esp32. It seems everything is correct, but I can't figure out why. I pretty sure the if info of my MQTT server is filled in correctly. Could you please help me with this? Thank you very much!

Below is the message from serial monitor:
`Initialize BLE client...
Connecting to WiFi...
...
WiFi connected

Connecting to MQTT...
MQTT connected

Processing Flora device at c4:7c:8d:62:37:e9 (try 1)
- Connection successful
- Found data service
- Force device in data mode
- Access characteristic from device

- Read value from characteristic
Hex: EC 0 0 D9 0 0 0 2E 3E 2 0 0 0 0 0 0
-- Temperature: 23.60
-- Moisture: 46
-- Light: 217
-- Conductivity: 574
WiFi disonnected
MQTT disconnected
Going to sleep now.`

sidddy, 2019-07-11:
that's weird, the log does not show any issue on esp32 side... does the problem still exist?

roc634, 2019-07-12:
Hi sidddy, thanks for your reply. I've tried several times including filling in the MQTT info letter by letter, just to make sure there is no mistake. Then I tried SubOrbit's project https://github.com/SubOrbit/flora  and it works well. So I stop looking for the reason. If it is only me reporting this issue, I think it must have been the mistake on my side, after all I'm a beginner. Thanks for your reply though. Have a nice day!

To: sidddy/flora
CC: roc634; Author
Subject: Re: [sidddy/flora] MQTT server received no message (#18)
that's weird, the log does not show any issue on esp32 side... does the problem still exist?

closed 2019-07-12 by sidddy