sidddy/flora #36: No Sleep / Restart. Need Live Data

canim60, 2020-09-05:
Hello is that possible to disable the restart of Wifi / MQT,??? I need the Live Data from the Flower Sensor, because i shower my Flowers with a Pump. If there Arrived Moisseture then will be off.  I Can set the Time from restart down, but its not ok for my logs, to many restart to MQT and Wifi Router. 
My Question is, is that possible to restart only the Bluetooth modul ? or any else.

iks73, 2020-10-30:
to solve your issue you have just to minimize SLEEP_DURATION in config.h
e.g. if you make
#define SLEEP_DURATION 10
the esp32 device will sleep only 10 seconds and then repeat to grab data from your sensors

canim60, 2020-11-01:
thx for reply, but there is to much connection to wifi and MQTT server

iks73, 2020-11-02:
try to use any server without those limitations, e.g. http://mqtt.by